nopnop2002/esp-idf-sx126x #2: Facing Error: RA01S: SPI Transaction error:10

testqaibs, 2022-04-14:
I am using tried using ESP32-WROOM-32D and ESP32-C3-13entry 0x40080694
I (26) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
I (26) boot: compile time 01:07:15
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot: End of partition table
I (93) boot_comm: chip revision: 1, min. application chip revision: 0
I (100) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0a394h ( 41876) map
I (124) esp_image: segment 1: paddr=0001a3bc vaddr=3ffb0000 size=02494h (  9364) load
I (128) esp_image: segment 2: paddr=0001c858 vaddr=40080000 size=037c0h ( 14272) load
I (136) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=196c0h (104128) map
I (176) esp_image: segment 4: paddr=000396e8 vaddr=400837c0 size=09070h ( 36976) load
I (192) esp_image: segment 5: paddr=00042760 vaddr=50000000 size=00010h (    16) load
I (198) boot: Loaded app from partition at offset 0x10000
I (198) boot: Disabling RNG early entropy source...
I (212) cpu_start: Pro cpu up.
I (212) cpu_start: Starting app cpu, entry point is 0x4008109c
0x4008109c: call_start_cpu1 at C:/Users/Pravin/esp/esp-idf/components/esp_system/port/cpu_start.c:156

I (0) cpu_start: App cpu up.
I (226) cpu_start: Pro cpu start user code
I (226) cpu_start: cpu freq: 160000000
I (226) cpu_start: Application information:
I (230) cpu_start: Project name:     lora
I (235) cpu_start: App version:      1
I (240) cpu_start: Compile time:     Apr 14 2022 01:06:41
I (246) cpu_start: ELF file SHA256:  d7058e82e5363bb4...
I (252) cpu_start: ESP-IDF:          v4.4-dirty
I (257) heap_init: Initializing. RAM available for dynamic allocation:
I (264) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (270) heap_init: At 3FFB2DD8 len 0002D228 (180 KiB): DRAM
I (276) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (283) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (289) heap_init: At 4008C830 len 000137D0 (77 KiB): IRAM
I (297) spi_flash: detected chip: generic
I (300) spi_flash: flash io: dio
I (305) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (315) main: Frequency is 433MHz
I (315) RA01S: CONFIG_MISO_GPIO=12
I (325) RA01S: CONFIG_MOSI_GPIO=13
I (325) RA01S: CONFIG_SCLK_GPIO=15
I (335) RA01S: CONFIG_NSS_GPIO=14
I (335) RA01S: CONFIG_RST_GPIO=33
I (335) RA01S: CONFIG_BUSY_GPIO=17
I (345) RA01S: CONFIG_TXEN_GPIO=-1
I (345) RA01S: CONFIG_RXEN_GPIO=-1
I (355) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (365) gpio: GPIO[33]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (375) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (375) RA01S: spi_bus_initialize=0
I (385) RA01S: spi_bus_add_device=0
I (1425) RA01S: Reset
I (3425) RA01S: syncWord=0x1424
I (3425) RA01S: SX126x installed
E (5425) RA01S: SPI Transaction error:10

nopnop2002, 2022-04-14:
>I am using tried using ESP32-WROOM-32D and ESP32-C3-13 devices

```
I (315) RA01S: CONFIG_MISO_GPIO=12
I (325) RA01S: CONFIG_MOSI_GPIO=13
I (325) RA01S: CONFIG_SCLK_GPIO=15
I (335) RA01S: CONFIG_NSS_GPIO=14
I (335) RA01S: CONFIG_RST_GPIO=33
I (335) RA01S: CONFIG_BUSY_GPIO=17
```

ESP32-C3 don't have GPIO12,13,14,15,33,17

testqaibs, 2022-04-14:
Thanks for the quick reply. I have enabled LoRaDebugPrint, now I am getting following logs:

I (375) RA01S: spi_bus_initialize=0
I (385) RA01S: spi_bus_add_device=0
I (1425) RA01S: Reset
I (2425) RA01S: ReadRegister:   REG=0x740
I (2425) RA01S: DataIn:14 
I (2425) RA01S: DataIn:24 
I (3425) RA01S: syncWord=0x1424
I (3425) RA01S: SX126x installed
I (4425) RA01S: WriteCommand:   CMD=0x80
I (4425) RA01S: 00 --> aa
E (5425) RA01S: SPI Transaction error:10

nopnop2002, 2022-04-14:
>I (315) main: Frequency is 433MHz

Are you using RA-01S?

testqaibs, 2022-04-14:
Thanks for giving your logs for comparison and timely replies.
I am using LoRAa1262F30-915 chipset i.e from G-NiceRF. Hope this chipset is also supported with this code.

nopnop2002, 2022-04-14:
I don't know LoRAa1262F30.

Is the data sheet published?

testqaibs, 2022-04-15:
https://www.aliexpress.com/item/4000415800912.html

nopnop2002, 2022-04-15:
I want to see the schematic inside the LoRa1262F30
Without seeing this, the following cannot be judged.
```
Using TCXO(Temperature-Compensated Crystal Oscillator)
SX1262/1268 can use the TCXO.
If the TCXO is used, the XTB pin is not connected.
However, the 6th pin (DIO3) of the SX1262/1268 can be used to power the TCXO.
Explanation for TXCO and antenna control is [here](https://github.com/beegee-tokyo/SX126x-Arduino).
Ra-01S / Ra-01SH does not use TCXO.

Power supply modes
SX1262/1268 has two power supply modes.
One is that only LDO used in all modes.
Another is that DC_DC+LDO used for STBY_XOSC,FS, RX and TX modes.
Explanation for LDO and DCDC selection is [here](https://github.com/beegee-tokyo/SX126x-Arduino).
Ra-01S / Ra-01SH use only LDO in all modes.

RF-Switching
In general, use DIO2 to switch the RF-Switch.
However, some tranceiver use an external gpio to switch the RF-Switch.
Ra-01S / Ra-01SH use the SC70-6 integrated load switch to switch between RFO and RFI.
Ra-01S / Ra-01SH use DIO2 to control this.
DIO2 = 1, CTRL = 0, RFC to RF1 Tx Mode.
DIO2 = 0, CTRL = 1, RFC to R21 Rx Mode.
```

nopnop2002, 2022-04-15:
This is Ra-01S internal schematic.

If you look at this, you can see the following.

Ra-01S / Ra-01SH does not use TCXO.
Ra-01S / Ra-01SH does not use TCXO.
Ra-01S / Ra-01SH use only LDO in all modes.
Ra-01S / Ra-01SH use the SC70-6 integrated load switch to switch between RFO and RFI.
Ra-01S / Ra-01SH use DIO2 to control this.

[image]

testqaibs, 2022-04-15:
https://www.nicerf.com/products/detail/sx1262-lora-module-lora1262f30.html

nopnop2002, 2022-04-15:
I saw Product Spec.

If tcxoVoltage is set to a value other than 0, TCXO will be used, but I don't know what value is appropriate.
The Product Spec does not mention this.

If useRegulatorLDO is set to true, DC_DC+LDO used for STBY_XOSC,FS, RX and TX modes.
The Product Spec does not mention this.

```
	float tcxoVoltage = 0.0; // don't use TCXO
	bool useRegulatorLDO = false; // use only LDO in all modes
	int ret = LoRaBegin(frequencyInHz, txPowerInDbm, tcxoVoltage, useRegulatorLDO);
```

nopnop2002, 2022-05-24:
No.

The appropriate value for tcxoVoltage is not listed in the datasheet.

mx-cyber, 2022-05-24:
OK, thank you very much for your reply

Saba-abiri, 2025-03-04:
Hi [mx-cyber](https://github.com/mx-cyber),[nopnop2002](https://github.com/nopnop2002) and [testqaibs](https://github.com/testqaibs)

I had this issue too and I fixed it!
first of all before any command do it :

set NRESET_LORA to Low
wait for about 200 milli seconds
set NRESET_LORA to High
wait for about 500 milli seconds

then send this command

// SetDIO3asTCXOCtrl SPI Transaction
cmnd[0] = 0x97; // 0x97 is SetDIO3asTCXOCtrl
cmnd[1] = 0x00; // DIO3 outputs 1.6 V to supply the TCXO
cmnd[2] = 0x00;
cmnd[3] = 0x00;

let me know if you fix it 👍

(base on my repository you can see more setting : https://github.com/Saba-abiri/LORA1262/blob/main/LORA1262/Libs/Src/SX1262.c)

nopnop2002, 2025-03-04:
@Saba-abiri 

Try SetDio3AsTcxoCtrl(0.0,9);

Saba-abiri, 2025-03-04:
[nopnop2002](https://github.com/nopnop2002)

right after reset the sx1262 you have a very short time to set tcxo voltage!

if you do it after that ,it won't work!

so the first thing when sx1262 reset is to set tcxo voltage.
(actually it is a bug in lora1262f30 and this is how to fix it so don't use SetDio3AsTcxoCtrl(0.0,0); after all config , do it before config directly)