nopnop2002/esp-idf-sx126x #4: Ebyte E22-900M22S: SPI Transaction error:10

joaoceresoli, 2022-07-05:
I have a same issue that previous, but in a different module.
I use a Ebyte e22-900m22s module, that have a LoRa chip llcc68, its seems a semtech 126x.
The data from this devices is [here](https://www.ebyte.com/en/product-view-news.html?id=1360) and [here](https://www.mouser.com/pdfDocs/DS_LLCC68_V10-2.pdf).
Output from my log.

> I (0) cpu_start: App cpu up.
> I (226) cpu_start: Pro cpu start user code
> I (226) cpu_start: cpu freq: 160000000
> I (226) cpu_start: Application information:
> I (231) cpu_start: Project name:     lora
> I (236) cpu_start: App version:      09d3831
> I (241) cpu_start: Compile time:     Jul  4 2022 19:21:46
> I (247) cpu_start: ELF file SHA256:  e0e3043ee90ffc9d...
> I (253) cpu_start: ESP-IDF:          v4.4-dirty
> I (258) heap_init: Initializing. RAM available for dynamic allocation:
> I (265) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
> I (271) heap_init: At 3FFB2DD8 len 0002D228 (180 KiB): DRAM
> I (277) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
> I (284) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
> I (290) heap_init: At 4008C824 len 000137DC (77 KiB): IRAM
> I (298) spi_flash: detected chip: generic
> I (301) spi_flash: flash io: dio
> I (306) cpu_start: Starting scheduler on PRO CPU.
> I (0) cpu_start: Starting scheduler on APP CPU.
> I (316) main: Frequency is 433MHz
> I (316) RA01S: CONFIG_MISO_GPIO=19
> I (326) RA01S: CONFIG_MOSI_GPIO=23
> I (326) RA01S: CONFIG_SCLK_GPIO=18
> I (336) RA01S: CONFIG_NSS_GPIO=5
> I (336) RA01S: CONFIG_RST_GPIO=14
> I (336) RA01S: CONFIG_BUSY_GPIO=26
> I (346) RA01S: CONFIG_TXEN_GPIO=-1
> I (346) RA01S: CONFIG_RXEN_GPIO=-1
> I (356) gpio: GPIO[5]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (366) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (376) gpio: GPIO[26]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
> I (376) RA01S: spi_bus_initialize=0
> I (386) RA01S: spi_bus_add_device=0
> I (426) RA01S: Reset
> I (426) RA01S: ReadRegister: REG=0x740
> I (426) RA01S: DataIn:14 
> I (426) RA01S: DataIn:24 
> I (436) RA01S: syncWord=0x1424
> I (436) RA01S: SX126x installed
> I (436) RA01S: WriteCommand: CMD=0x80
> I (446) RA01S: 00 --> aa
> E (446) RA01S: SPI Transaction error:10

### I decided to remove the while(1) lock for when an error occurs, in order to check if all registers generated errors, I noticed that not all of them, here is my output:

> I (426) RA01S: Reset
> I (426) RA01S: ReadRegister: REG=0x740
> I (426) RA01S: DataIn:14 
> I (426) RA01S: DataIn:24 
> I (436) RA01S: syncWord=0x1424
> I (436) RA01S: SX126x installed
> I (436) RA01S: WriteCommand: CMD=0x80
> I (446) RA01S: 00 --> aa
> E (446) RA01S: SPI Transaction error:10
> I (456) RA01S: WriteCommand: CMD=0x9d
> I (456) RA01S: 01 --> a2
> I (456) RA01S: tcxoVoltage=0.000000
> I (466) RA01S: WriteCommand: CMD=0x89
> I (466) RA01S: 7f --> a2
> I (476) RA01S: useRegulatorLDO=0
> I (476) RA01S: WriteCommand: CMD=0x96
> I (476) RA01S: 01 --> aa
> E (486) RA01S: SPI Transaction error:10
> I (486) RA01S: WriteCommand: CMD=0x8f
> I (486) RA01S: 00 --> a2
> I (496) RA01S: 00 --> a2
> I (496) RA01S: WriteCommand: CMD=0x95
> I (506) RA01S: 04 --> a2
> I (506) RA01S: 07 --> a2
> I (506) RA01S: 00 --> a2
> I (516) RA01S: 01 --> a2
> I (516) RA01S: WriteRegister: REG=0x8e7
> I (516) RA01S: 18 --> a2
> I (526) RA01S: WriteCommand: CMD=0x8e
> I (526) RA01S: 16 --> a2
> I (526) RA01S: 04 --> a2
> I (536) RA01S: WriteCommand: CMD=0x98
> I (536) RA01S: 6b --> a2
> I (536) RA01S: 6f --> a2
> I (546) RA01S: WriteCommand: CMD=0x86
> I (546) RA01S: 1b --> aa
> E (556) RA01S: SPI Transaction error:10
> I (556) MAIN: LoRaBegin=0
> I (556) RA01S: SetStopRxTimerOnPreambleDetect enable=0
> I (566) RA01S: WriteCommand: CMD=0x9f
> I (566) RA01S: 00 --> a2
> I (576) RA01S: WriteCommand: CMD=0xa0
> I (576) RA01S: 00 --> a2
> I (576) RA01S: WriteCommand: CMD=0x8a
> I (586) RA01S: 01 --> a2
> I (586) RA01S: WriteCommand: CMD=0x8b
> I (596) RA01S: 07 --> a2
> I (596) RA01S: 04 --> a2
> I (596) RA01S: 01 --> a2
> I (606) RA01S: 00 --> a2
> I (606) RA01S: ReadRegister: REG=0x736
> I (606) RA01S: DataIn:0d 
> I (616) RA01S: WriteRegister: REG=0x736
> I (616) RA01S: 09 --> a2
> I (616) RA01S: WriteCommand: CMD=0x8c
> I (626) RA01S: 00 --> a2
> I (626) RA01S: 08 --> a2
> I (626) RA01S: 00 --> a2
> I (636) RA01S: ff --> a2
> I (636) RA01S: 01 --> a2
> I (636) RA01S: 00 --> a2
> I (646) RA01S: WriteCommand: CMD=0x08
> I (646) RA01S: 03 --> a2
> I (656) RA01S: ff --> a2
> I (656) RA01S: 00 --> a2
> I (656) RA01S: 00 --> a2
> I (656) RA01S: 00 --> a2
> I (666) RA01S: 00 --> a2
> I (666) RA01S: 00 --> a2
> I (666) RA01S: 00 --> a2
> I (676) RA01S: ----- SetRx timeout=16777215
> I (676) RA01S: WriteCommand: CMD=0x80
> I (686) RA01S: 00 --> a2
> I (686) RA01S: SetRxEnable:SX126x_TXEN=-1 SX126x_RXEN=-1
> I (696) RA01S: WriteCommand: CMD=0x82
> I (696) RA01S: ff --> a2
> I (696) RA01S: ff --> a2
> I (706) RA01S: ff --> a2
> I (706) RA01S: ReadCommand: CMD=0xc0
> I (706) RA01S: DataIn:2a
> I (716) RA01S: ReadCommand: CMD=0xc0
> I (716) RA01S: DataIn:2a
> I (726) RA01S: ReadCommand: CMD=0xc0
> I (726) RA01S: DataIn:2a
> I (726) RA01S: ReadCommand: CMD=0xc0
> I (736) RA01S: DataIn:2a
> I (736) RA01S: ReadCommand: CMD=0xc0
> I (736) RA01S: DataIn:2a
> I (746) RA01S: ReadCommand: CMD=0xc0
> I (746) RA01S: DataIn:2a
> I (756) RA01S: ReadCommand: CMD=0xc0
> I (756) RA01S: DataIn:2a
> I (756) RA01S: ReadCommand: CMD=0xc0
> I (766) RA01S: DataIn:2a
> I (766) RA01S: ReadCommand: CMD=0xc0
> I (766) RA01S: DataIn:2a
> I (776) RA01S: ReadCommand: CMD=0xc0
> I (776) RA01S: DataIn:2a
> I (786) RA01S: ReadCommand: CMD=0xc0
> I (786) RA01S: DataIn:2a
> E (786) RA01S: SetRx Illegal Status

 I noticed that the possible problem for the previous question was in tcxoVoltage, however I didn't understand how this parameter works and what value exactly I have to configure.

joaoceresoli, 2022-07-05:
I find a possible solution. 
After testing using a arduino lib compatible with my module, i found tcxo Voltage is 3.3v.
Even making this change, still not succeeding in putting the module on standby, however I implemented a routine for it to try again if the command returns failure and apparently solved the problem.

https://github.com/nopnop2002/esp-idf-sx126x/blob/09d3831df87559359bbacdcb3c58d17c510b22a7/components/ra01s/ra01s.c#L935

	for (uint8_t retry = 1; (in == 0xaa) && (retry < 10); retry++)
		{
			gpio_set_level(SX126x_SPI_SELECT, HIGH);
			WaitForIdle(BUSY_WAIT);
			gpio_set_level(SX126x_SPI_SELECT, LOW);
			spi_transfer(cmd);
			in = spi_transfer(data[n]);

			if (debugPrint)
			{
				ESP_LOGW(TAG, "RETRY: Try=%d", retry);
				ESP_LOGW(TAG, "RETRY:WriteCommand: CMD=0x%02x", cmd);
				ESP_LOGW(TAG, "RETRY:%02x --> %02x", data[n], in);
			}
		}

After this return is for me

> I (2462) RA01S: spi_bus_add_device=0
> I (2502) RA01S: Reset
> I (2502) RA01S: ReadRegister: REG=0x740
> I (2502) RA01S: DataIn:14 
> I (2502) RA01S: DataIn:24 
> I (2512) RA01S: syncWord=0x1424
> I (2512) RA01S: SX126x installed
> I (2512) RA01S: WriteCommand: CMD=0x80
> I (2522) RA01S: 00 --> aa
> W (2522) RA01S: RETRY: Try=1
> W (2532) RA01S: RETRY:WriteCommand: CMD=0x80
> W (2532) RA01S: RETRY:00 --> a2
> I (2532) RA01S: WriteCommand: CMD=0x9d
> I (2542) RA01S: 01 --> a2
> I (2542) RA01S: tcxoVoltage=3.300000
> I (2552) RA01S: WriteCommand: CMD=0x97
> I (2552) RA01S: 07 --> a2
> I (2552) RA01S: 00 --> a2
> I (2562) RA01S: 01 --> a2
> I (2562) RA01S: 40 --> a2
> I (2562) RA01S: WriteCommand: CMD=0x89
> I (2572) RA01S: 7f --> a2
> I (2582) RA01S: useRegulatorLDO=0
> I (2582) RA01S: WriteCommand: CMD=0x96
> I (2582) RA01S: 01 --> a2
> I (2582) RA01S: WriteCommand: CMD=0x8f
> I (2592) RA01S: 00 --> a2
> I (2592) RA01S: 00 --> a2
> I (2592) RA01S: WriteCommand: CMD=0x95
> I (2602) RA01S: 04 --> a2
> I (2602) RA01S: 07 --> a2
> I (2612) RA01S: 00 --> a2
> I (2612) RA01S: 01 --> a2
> I (2612) RA01S: WriteRegister: REG=0x8e7
> I (2622) RA01S: 18 --> a2
> I (2622) RA01S: WriteCommand: CMD=0x8e
> I (2622) RA01S: 16 --> a2
> I (2632) RA01S: 04 --> a2
> I (2632) RA01S: WriteCommand: CMD=0x98
> I (2642) RA01S: e1 --> a2
> I (2642) RA01S: e9 --> a2
> I (2652) RA01S: WriteCommand: CMD=0x86
> I (2652) RA01S: 39 --> a2
> I (2652) RA01S: 30 --> a2
> I (2652) RA01S: 00 --> a2
> I (2662) RA01S: 00 --> a2
> I (2662) RA01S: SetStopRxTimerOnPreambleDetect enable=0
> I (2672) RA01S: WriteCommand: CMD=0x9f
> I (2672) RA01S: 00 --> a2
> I (2682) RA01S: WriteCommand: CMD=0xa0
> I (2682) RA01S: 00 --> a2
> I (2692) RA01S: WriteCommand: CMD=0x8a
> I (2692) RA01S: 01 --> a2
> I (2692) RA01S: WriteCommand: CMD=0x8b
> I (2702) RA01S: 07 --> a2
> I (2702) RA01S: 07 --> a2
> I (2702) RA01S: 01 --> a2
> I (2712) RA01S: 00 --> a2
> I (2712) RA01S: ReadRegister: REG=0x736
> I (2722) RA01S: DataIn:0d 
> I (2722) RA01S: WriteRegister: REG=0x736
> I (2722) RA01S: 09 --> a2
> I (2732) RA01S: WriteCommand: CMD=0x8c
> I (2732) RA01S: 00 --> a2
> I (2732) RA01S: 08 --> a2
> I (2742) RA01S: 00 --> a2
> I (2742) RA01S: ff --> a2
> I (2742) RA01S: 01 --> a2
> I (2752) RA01S: 00 --> a2
> I (2752) RA01S: WriteCommand: CMD=0x08
> I (2762) RA01S: 03 --> a2
> I (2762) RA01S: ff --> a2
> I (2762) RA01S: 00 --> a2
> I (2772) RA01S: 00 --> a2
> I (2772) RA01S: 00 --> a2
> I (2772) RA01S: 00 --> a2
> I (2782) RA01S: 00 --> a2
> I (2782) RA01S: 00 --> a2
> I (2782) RA01S: ----- SetRx timeout=16777215
> I (2792) RA01S: WriteCommand: CMD=0x80
> I (2792) RA01S: 00 --> a2
> I (2792) RA01S: SetRxEnable:SX126x_TXEN=-1 SX126x_RXEN=-1
> I (2802) RA01S: WriteCommand: CMD=0x82
> I (2802) RA01S: ff --> a2
> I (2812) RA01S: ff --> a2
> I (2812) RA01S: ff --> a2
> I (2822) RA01S: ReadCommand: CMD=0xc0
> I (2822) RA01S: DataIn:52
> I (2822) RA01S: ReadCommand: CMD=0xc0
> I (2832) RA01S: DataIn:52

### Now I'm going to test if I can effectively communicate.

nopnop2002, 2022-07-05:
What if you add a little wait here?

```
I (2512) RA01S: SX126x installed

Wait for few second.

I (2512) RA01S: WriteCommand: CMD=0x80
I (2522) RA01S: 00 --> aa
W (2522) RA01S: RETRY: Try=1
W (2532) RA01S: RETRY:WriteCommand: CMD=0x80
W (2532) RA01S: RETRY:00 --> a2
I (2532) RA01S: WriteCommand: CMD=0x9d
I (2542) RA01S: 01 --> a2
```

joaoceresoli, 2022-07-06:
Ironically it now works without adding delay and without going through my retry loop.
as you say, these modules are quite esoteric....
I'm trying to communicate with the sx176x modules, in this case the heltec devkit boards, but I still haven't had success in communication, I believe it's a matter of the lora settings, as there are several parameters.

nopnop2002, 2022-07-06:
>these modules are quite esoteric....

I think so, too.

joaoceresoli, 2022-07-06:
Updates:
I managed to communicate with the heltec boards perfectly, the lora is working!!
On that other issue of adding the delay, I had forgotten to enable the loradebug mode, so it looked like the error was gone, but it didn't, and even adding it still persists.
i have add a delay(30000) [before this line](https://github.com/nopnop2002/esp-idf-sx126x/blob/09d3831df87559359bbacdcb3c58d17c510b22a7/components/ra01s/ra01s.c#L180)
> I (5582) RA01S: WriteCommand: CMD=0x80
> I (5582) RA01S: 00 --> aa
> W (5582) RA01S: RETRY: Try=1
> W (5582) RA01S: RETRY:WriteCommand: CMD=0x80
> W (5592) RA01S: RETRY:00 --> a2

### Checking the rest of the log, it generates this retry only this first time when entering SetStandby(), other times it enters normally.

netmonk, 2022-07-23:
Interesting i was about  to post the same issue here.i face the same problem today   trying the lib with an esp32-c3 and a e22 900m22s module.

nopnop2002, 2022-07-24:
@netmonk 

https://github.com/nopnop2002/esp-idf-sx126x/blob/09d3831df87559359bbacdcb3c58d17c510b22a7/components/ra01s/ra01s.c#L935

netmonk, 2022-07-24:
I was searching for an 'if' to test the previous cmd status to enter the loop but  in fact  it si the loop condition itself. great 

Now my dump  of execution is :
```
I (565) RA01S: spi_bus_add_device=0                
I (615) RA01S: Reset                               
I (615) RA01S: ReadRegister: REG=0x740             
I (615) RA01S: DataIn:14             
I (615) RA01S: DataIn:24                           
I (615) RA01S: syncWord=0x1424       
I (615) RA01S: SX126x installed                    
I (625) RA01S: WriteCommand: CMD=0x80
I (625) RA01S: 00 --> aa                           
W (635) RA01S: RETRY: Try=1                        
W (635) RA01S: RETRY:WriteCommand: CMD=0x80
W (635) RA01S: RETRY:00 --> a2
I (645) RA01S: WriteCommand: CMD=0x9d                                                                 
I (645) RA01S: 01 --> a2                           
I (645) RA01S: tcxoVoltage=0.000000  
I (655) RA01S: WriteCommand: CMD=0x89                                                                 
I (655) RA01S: 7f --> a2                           
I (665) RA01S: useRegulatorLDO=0                                                                      
I (665) RA01S: WriteCommand: CMD=0x96
I (665) RA01S: 01 --> aa
W (675) RA01S: RETRY: Try=1
W (675) RA01S: RETRY:WriteCommand: CMD=0x96
W (685) RA01S: RETRY:01 --> a2
I (685) RA01S: WriteCommand: CMD=0x8f
I (685) RA01S: 00 --> a2
I (695) RA01S: 00 --> a2
I (695) RA01S: WriteCommand: CMD=0x95
I (705) RA01S: 04 --> a2
I (705) RA01S: 07 --> a2
I (705) RA01S: 00 --> a2
I (705) RA01S: 01 --> a2
I (715) RA01S: WriteRegister: REG=0x8e7
I (715) RA01S: 18 --> a2
I (725) RA01S: WriteCommand: CMD=0x8e
I (725) RA01S: 16 --> a2
I (725) RA01S: 04 --> a2
I (735) RA01S: WriteCommand: CMD=0x98
I (735) RA01S: d7 --> a2
I (735) RA01S: d8 --> a2
I (745) RA01S: WriteCommand: CMD=0x86
I (745) RA01S: 36 --> aa
W (745) RA01S: RETRY: Try=1
W (755) RA01S: RETRY:WriteCommand: CMD=0x86
W (755) RA01S: RETRY:36 --> a2
I (765) RA01S: 20 --> a2
I (765) RA01S: 00 --> a2
I (765) RA01S: 00 --> a2
I (775) MAIN: LoRaBegin=0 
I (775) RA01S: SetStopRxTimerOnPreambleDetect enable=0
I (785) RA01S: WriteCommand: CMD=0x9f
I (785) RA01S: 00 --> a2
I (785) RA01S: WriteCommand: CMD=0xa0
I (795) RA01S: 00 --> a2
I (795) RA01S: WriteCommand: CMD=0x8a
I (795) RA01S: 01 --> a2
I (805) RA01S: WriteCommand: CMD=0x8b
I (805) RA01S: 07 --> a2
I (815) RA01S: 04 --> a2
I (815) RA01S: 01 --> a2
I (815) RA01S: 00 --> a2
I (825) RA01S: ReadRegister: REG=0x736
I (825) RA01S: DataIn:0d  
I (825) RA01S: WriteRegister: REG=0x736
I (835) RA01S: 09 --> a2
I (835) RA01S: WriteCommand: CMD=0x8c
I (845) RA01S: 00 --> a2
I (845) RA01S: 08 --> a2
I (845) RA01S: 00 --> a2
I (845) RA01S: ff --> a2
I (855) RA01S: 01 --> a2
I (855) RA01S: 00 --> a2
I (855) RA01S: WriteCommand: CMD=0x08
I (865) RA01S: 03 --> a2
I (865) RA01S: ff --> a2
I (875) RA01S: 00 --> a2
I (875) RA01S: 00 --> a2
I (875) RA01S: 00 --> a2
I (875) RA01S: 00 --> a2
I (885) RA01S: 00 --> a2
I (885) RA01S: 00 --> a2
I (885) RA01S: ----- SetRx timeout=16777215
I (895) RA01S: WriteCommand: CMD=0x80
I (895) RA01S: 00 --> a2
I (905) RA01S: SetRxEnable:SX126x_TXEN=-1 SX126x_RXEN=-1
I (905) RA01S: WriteCommand: CMD=0x82
I (915) RA01S: ff --> a2
I (915) RA01S: ff --> a2
I (915) RA01S: ff --> a2
I (925) RA01S: ReadCommand: CMD=0xc0
I (925) RA01S: DataIn:2a
I (935) RA01S: ReadCommand: CMD=0xc0
I (935) RA01S: DataIn:2a
I (935) RA01S: ReadCommand: CMD=0xc0
I (945) RA01S: DataIn:2a
I (945) RA01S: ReadCommand: CMD=0xc0
I (945) RA01S: DataIn:2a
I (955) RA01S: ReadCommand: CMD=0xc0
I (955) RA01S: DataIn:2a
I (965) RA01S: ReadCommand: CMD=0xc0
I (965) RA01S: DataIn:2a
I (965) RA01S: ReadCommand: CMD=0xc0
I (975) RA01S: DataIn:2a
I (975) RA01S: ReadCommand: CMD=0xc0
I (975) RA01S: DataIn:2a
I (985) RA01S: ReadCommand: CMD=0xc0
I (985) RA01S: DataIn:2a
I (985) RA01S: ReadCommand: CMD=0xc0
I (995) RA01S: DataIn:2a
I (995) RA01S: ReadCommand: CMD=0xc0
I (1005) RA01S: DataIn:2a 
E (1005) RA01S: SetRx Illegal Status
```
by the way where can i find error code table like what 'aa' means nd what 'a2' means ?

netmonk, 2022-07-24:
@joaoceresoli  from my understanding `e220 900m22s` and `e22 900m22s` are difeent module the first one is based on new lora chip `llcc68` and the second one is based on `sx1261/62` chip.  Therefore  the user manual are different.

In the issue tittle you eference `e22 900m22s` but in your first post you link to `llcc68` datasheet.
You can find the `sx1261/62` user manualx here https://github.com/netmonk/mcudoc/tree/master/lorachip

netmonk, 2022-07-24:
[image]
Do you remenber this port expension board lost on the way to Japan ? @nopnop2002 have a good laugh :)

nopnop2002, 2022-07-24:
>by the way where can i find error code table like what aa means nd what a2 means ?

As far as I can tell, status bytes is almost always returned when the microcontroller is sending data to the SX126x.

[image]

0xA2 is STBY_RC + RFU
0xAA is STBY_RC + Failure to execute command

[image]

joaoceresoli, 2022-07-25:
> @joaoceresoli from my understanding `e220 900m22s` and `e22 900m22s` are different module the first one is based on new lora chip `llcc68` and the second one is based on `sx1261/62` chip. Therefore the user manual are different.
> 
> In the issue tittle you reference `e22 900m22s` but in your first post you link to `llcc68` datasheet. You can find the `sx1261/62` user manualx here https://github.com/netmonk/mcudoc/tree/master/lorachip

Really, i'm sorry, you are correct, my lora module has a semtech sx126x, not a llcc... I remember reading that the registers of the sx126x and llcc68 modules were the same, the evolution of the chip would be in the part of improvements in the RF circuit, but I could be wrong.

nopnop2002, 2022-07-26:
LLCC68 is compatible with SX126X.
However, LLCC68 has the following restrictions:

BW is either 125KHz, 250KHz or 500Khz.
When BW is 125KHz, SF is in the range of 5-9.
When BW is 250KHz, SF is in the range of 5-10.
When BW is 500KHz, SF is in the range of 5-11.

netmonk, 2022-07-26:
From what I know sf 5 and sf 6 are not compatible with sx127*

netmonk, 2022-07-26:
> This repository works correctly with Ra-01SC (LLCC68).

You should rename the repo then :)

netmonk, 2022-07-27:
can someone explain to me the following test on getstatus() ? 
```
681          for(int retry=0;retry<10;retry++) {                                                                                                                                                               
   1                 if ((GetStatus() & 0x70) == 0x50) break;                                                                                                                                                  
   2                 delay(1);                                      
```

i dont understand the logic to test first 01110000 to test 01010000 at the end cause i dont see what  0111 is masking

nopnop2002, 2022-07-27:
This confirms whether the chip mode is RX.

https://github.com/nopnop2002/esp-idf-sx126x/blob/main/components/ra01s/ra01s.c#L681

netmonk, 2022-07-28:
and why not test like this :  if ((GetStatus() & 0x50) == 0x50) break; ? 
using 0x50 instead of 0x70 ?

nopnop2002, 2022-07-28:
This is same.

if ((GetStatus() & 0x70) == 0x50) break; 

if ((GetStatus() & 0x50) == 0x50) break;

netmonk, 2022-07-28:
The module is not entering rx state it return 2a

joaoceresoli, 2022-07-28:
@netmonk  Give my a some time to adjust my code for example model.

joaoceresoli, 2022-07-28:
You can read the modification i have made for module E22-900M22S work.
A litle detail about this module is TCXO voltage 3.3 is needed, i have altereted in example code.
link to [fork](https://github.com/joaoceresoli/esp-idf-sx126x).

joaoceresoli, 2022-07-28:
Other detail is my module is connected to a different pins in MCU, please make attention to it because i have altered kconfig.

netmonk, 2022-07-30:
thank you @joaoceresoli it works, it think the main point was about powering the tcxo. 
From my understanding, i was only looking in the library to manage this, while in fact it was in tthe main program

i guess it's time to adapt the library to support differents modules, by itself..

nopnop2002, 2022-08-01:
@joaoceresoli 
@netmonk 

Please try this and see if it works correctly.
``
	LoRaInit();
	//int ret = LoRaBegin(915000000, 22, 0.0, false);
	int8_t txPowerInDbm = 22;
	//float tcxoVoltage = 0.0; // don't use TCXO
	//bool useRegulatorLDO = false; // use only LDO in all modes

	float tcxoVoltage = 3.3; // use TCXO
	bool useRegulatorLDO = true; // use TCXO

	//LoRaDebugPrint(true);
	int ret = LoRaBegin(frequencyInHz, txPowerInDbm, tcxoVoltage, useRegulatorLDO);
	ESP_LOGI(TAG, "LoRaBegin=%d", ret);
```

netmonk, 2022-08-01:
For me, I will try to investigate with a oscilloscope the mosi pin during write transaction. There can be some error of pull-up/pull-down

nopnop2002, 2022-08-01:
>I will try to investigate with a oscilloscope the mosi pin during write transaction. 

It's good way.