nopnop2002/esp-idf-sx126x #8: E (456) RA01S: SetRx Illegal Status

niubbbbbbbb, 2022-10-22:
Hello, how to solve this problem.

nopnop2002, 2022-10-22:
If that doesn't work, enable [LoRaDebugPrint](https://github.com/nopnop2002/esp-idf-sx126x/blob/main/basic/main/main.c#L101).

niubbbbbbbb, 2022-10-23:
Do you play telegram or wechat? Can you add your friends to discuss it?

nopnop2002, 2022-10-23:
>Do you play telegram or wechat?

No

niubbbbbbbb, 2022-10-23:
😂😂All right.My English is not good either

niubbbbbbbb, 2022-10-23:
I think the log is not disabled. I'll show you my log. Please analyze it. I think it may be that I regret an error value after writing the command
[image]

nopnop2002, 2022-10-23:
0x86 is Set the RF frequency of the radio.

You may be using the wrong frequency.

RA-01S is 410-525Mhz.

closed 2022-10-23 by nopnop2002