nopnop2002/esp-idf-sx126x #9: I (737) RA01S: ReadCommand: CMD=0x12 I (737) RA01S: DataIn:d2 I (747) RA01S: DataIn:00 I (747) RA01S: DataIn:00

niubbbbbbbb, 2022-10-23:
I'm sorry to bother you again, but I have a question. Why can't I receive messages after setting successfully? My other Lora is sending messages normally.
[image]
[image]

niubbbbbbbb, 2022-10-23:
[image]

nopnop2002, 2022-10-23:
Can't you communicate between RA01S?

niubbbbbbbb, 2022-10-23:
I found the reason. It may be that the initialization of iqConfigCurrent needs to be modified due to the manufacturer's secondary modification. Now I can send and receive. Thank you for sharing the code and answering

nopnop2002, 2022-10-23:
Are you using AI-Thinker RA-01S?

niubbbbbbbb, 2022-10-23:
void FixInvertedIQ(uint8_t iqConfig)
{
	// fixes IQ configuration for inverted IQ
	// see SX1262/SX1268 datasheet, chapter 15 Known Limitations, section 15.4 for details
	// When exchanging LoRa packets with inverted IQ polarity, some packet losses may be observed for longer packets.
	// Workaround: Bit 2 at address 0x0736 must be set to:
	// “0” when using inverted IQ polarity (see the SetPacketParam(...) command)
	// “1” when using standard IQ polarity

	// read current IQ configuration
	uint8_t iqConfigCurrent = 0;
	ReadRegister(SX126X_REG_IQ_POLARITY_SETUP, &iqConfigCurrent, 1); // 0x0736

	set correct IQ configuration
	// if(iqConfig == SX126X_LORA_IQ_STANDARD) {   ///< last
	// 	iqConfigCurrent &= 0xFB;
	// } else {
	// 	iqConfigCurrent |= 0x04;
	// }

	if(iqConfig == SX126X_LORA_IQ_STANDARD) {   ///< now
		iqConfigCurrent &= 0xFB;
	} else {
		iqConfigCurrent |= 0x04;
	}

	// iqConfigCurrent = iqConfigCurrent | ( 1 << 2 );

	// update with the new value
	WriteRegister(SX126X_REG_IQ_POLARITY_SETUP, &iqConfigCurrent, 1); // 0x0736
}

niubbbbbbbb, 2022-10-23:
I use RA-01SH, that is, SX1262

nopnop2002, 2022-10-23:
This is original code.

I think there is no changes.

```
void FixInvertedIQ(uint8_t iqConfig)
{
    // fixes IQ configuration for inverted IQ
    // see SX1262/SX1268 datasheet, chapter 15 Known Limitations, section 15.4 for details
    // When exchanging LoRa packets with inverted IQ polarity, some packet losses may be observed for longer packets.
    // Workaround: Bit 2 at address 0x0736 must be set to:
    //  0”when using inverted IQ polarity (see the SetPacketParam(...) command)
    //  1”when using standard IQ polarity

    // read current IQ configuration
    uint8_t iqConfigCurrent = 0;
    ReadRegister(SX126X_REG_IQ_POLARITY_SETUP, &iqConfigCurrent, 1); // 0x0736

    // set correct IQ configuration
    if(iqConfig == SX126X_LORA_IQ_STANDARD) {
        iqConfigCurrent &= 0xFB;
    } else {
        iqConfigCurrent |= 0x04;
    }

    // update with the new value
    WriteRegister(SX126X_REG_IQ_POLARITY_SETUP, &iqConfigCurrent, 1); // 0x0736
}
```

niubbbbbbbb, 2022-10-24:
Sorry,I write it wrongly.If (iqConfig==SX126X_LORA_IQ_STANDARD) is replaced by if (iqConfig==SX126X_LORA_IQ_INVERTED)

nopnop2002, 2022-10-24:
>If (iqConfig==SX126X_LORA_IQ_STANDARD) is replaced by if (iqConfig==SX126X_LORA_IQ_INVERTED)

It seems to be a bug in the original code.

I'll fix it.

Thank you for reporting.

closed 2022-12-30 by niubbbbbbbb