nopnop2002/esp-idf-sx126x #10: Hello，Does the demo support CAD mode? Does LLCC68 support?

niubbbbbbbb, 2022-12-30:


nopnop2002, 2022-12-30:
What is  CAD mode? 

LLCC68 is used for Ra-01SC.   
LLCC68 is compatible with SX126X.   
However, there are the following restrictions:   
- BW is either 125KHz, 250KHz or 500Khz.   
- When BW is 125KHz, SF is in the range of 5-9.   
- When BW is 250KHz, SF is in the range of 5-10.   
- When BW is 500KHz, SF is in the range of 5-11.

niubbbbbbbb, 2022-12-30:
CAD is Channel Activity Detection

nopnop2002, 2022-12-30:
This library does not support CAD.

niubbbbbbbb, 2022-12-30:
Do you have support CAD's library?

niubbbbbbbb, 2022-12-30:
Can CAD functions be added to this library? Is there any problem in adding CAD functions to the library

nopnop2002, 2022-12-30:
there is no problem in adding CAD functions to the library.

closed 2022-12-30 by niubbbbbbbb

nopnop2002, 2023-01-05:
I added SetCadParams() and SetCad().

You can specify the following constants in SetCadParams().

```
//SX126X_CMD_SET_CAD_PARAMS
#define SX126X_CAD_ON_1_SYMB                          0x00        //  7     0     number of symbols used for CAD: 1
#define SX126X_CAD_ON_2_SYMB                          0x01        //  7     0                                     2
#define SX126X_CAD_ON_4_SYMB                          0x02        //  7     0                                     4
#define SX126X_CAD_ON_8_SYMB                          0x03        //  7     0                                     8
#define SX126X_CAD_ON_16_SYMB                         0x04        //  7     0                                     16
#define SX126X_CAD_GOTO_STDBY                         0x00        //  7     0     after CAD is done, always go to STDBY_RC mode
#define SX126X_CAD_GOTO_RX                            0x01        //  7     0     after CAD is done, go to Rx mode if activity is detected
```

At the end of the search period, the device triggers the IRQ CADdone if it has been enabled. 
If a valid signal has been detected it also generates the IRQ CadDetected.
These can be read using GetIrqStatus().

niubbbbbbbb, 2023-01-05:
That means you can use the CAD function directly, right

nopnop2002, 2023-01-05:
yes.
you are right.
If you use the CAD function, you need to read the datasheet.

niubbbbbbbb, 2023-01-05:
Thank you so much, brother