nopnop2002/esp-idf-sx126x #14: abort() istead of while(1) { vTaskDelay(1); }?

Glowman554, 2023-01-11:
Wouldn't it be better to use `abort()` or some other sort of error handler instead of `while(1) { vTaskDelay(1); }` when an error occurs? It would allow the user to handle errors or at least auto reboot the esp32.

nopnop2002, 2023-01-11:
You can auto-restart with esp_restart().

However, unless the error is recovered, it will auto-restart forever.

Glowman554, 2023-01-12:
For development purposes esp-idf allows you to turn auto restarts off. It allows you to have the feature when deploying.

nopnop2002, 2023-01-12:
>For development purposes esp-idf allows you to turn auto restarts off. 

how do i set it?

Glowman554, 2023-01-12:
Using CONFIG_ESP_SYSTEM_PANIC_PRINT_HALT halts the cpu instead of rebooting. I think it is located under components - > esp system settings

Glowman554, 2023-01-12:
The default configuration is CONFIG_ESP_SYSTEM_PANIC_PRINT_REBOOT

nopnop2002, 2023-01-12:
>Using CONFIG_ESP_SYSTEM_PANIC_PRINT_HALT halts the cpu instead of rebooting. I think it is located under components - > esp system settings

Thank you

Glowman554, 2023-01-13:
Hey would it be ok if i implement a error callback function which by default uses `while(1) { vTaskDelay(1); }` but can be overwritten by the user using weak linking?

nopnop2002, 2023-01-13:
>Hey would it be ok if i implement a error callback function which by default uses while(1) { vTaskDelay(1); } but can be overwritten by the user using weak linking?

Try it.

closed 2023-04-27 by Glowman554