nopnop2002/esp-idf-sx126x #19: Why is LLCC68 SetStandby(SX126X_STANDBY_RC) ; SPI Transaction error: 10

niubbbbbbbb, 2023-02-10:
Why is LLCC68 SetStandby (SX126X_STANDBY_RC) used; SPI Transaction error: 10

nopnop2002, 2023-02-10:
I can't understand what the problem is.

niubbbbbbbb, 2023-02-10:
When I use LLCC68, I used SetStandby (SX126X_STANDBY_RC) in LoRaBegin,but in SetStandby SPI Transaction error: 10

niubbbbbbbb, 2023-02-10:
[image]
[image]
LLCC68

nopnop2002, 2023-02-10:
Are there devices other than LLCC68 on the **same SPI bus?**

niubbbbbbbb, 2023-02-10:
No one on the same bus

nopnop2002, 2023-02-10:
I want to see all logging

niubbbbbbbb, 2023-02-10:
I didn't save the log, but I remember these values
[image]

nopnop2002, 2023-02-10:
Unfortunately, I can't analyze it without seeing all the logging.

niubbbbbbbb, 2023-02-10:
Waiting for the device to reconnect....
I (271) esp_image: segment 4: paddr=000c9adc vaddr=4037bcI (1388) nvm_dev: nvm dev create type = 1 result:0
I (1389) nvm_dev: TNvmDevDoOpen result = 0 op_state_ = 2
I (1389) nvm_dev: TNvmDevOpen type:1 result:0
AstPmmInit 
I (1390) pmm_bitmap: TPmmBitMapCreate - i=132096  8  0  0
I (1390) pmm_bitmap: TPmmBitMapLoad - 1 = 0x00000000
I (1390) nvm_like_eeprom: TNvmSdEepromRead 1  offset = 32768-32768 
I (1391) pmm_bitmap: TPmmBitMapLoad - 2 0x00000000  8
I (1391) nvm_like_eeprom: TNvmSdEepromRead 1  offset = 32776-32768 
I (1392) pmm_bitmap: TPmmBitMapCreate - i=133120  4  2048  0
I (1392) pmm_bitmap: TPmmBitMapLoad - 1 = 0x00000800
I (1393) nvm_like_eeprom: TNvmSdEepromRead 1  offset = 34816-32768 
I (1394) pmm_bitmap: TPmmBitMapLoad - 2 0x00000800  8
I (1394) nvm_like_eeprom: TNvmSdEepromRead 1  offset = 34824-32768 
I (1395) pmm_bitmap: TPmmBitMapCreate - i=134144  6  5120  0
I (1395) pmm_bitmap: TPmmBitMapLoad - 1 = 0x00001400
I (1395) nvm_like_eeprom: TNvmSdEepromRead 1  offset = 37888-36864 
I (1396) pmm_bitmap: TPmmBitMapLoad - 2 0x00001400  8
I (1396) nvm_like_eeprom: TNvmSdEepromRead 1  offset = 37896-36864 
I (1397) pmm_bitmap: TPmmBitMapCreate - i=135168  0  10240  0
I (1397) pmm_bitmap: TPmmBitMapLoad - 1 = 0x00002800
I (1398) nvm_like_eeprom: TNvmSdEepromRead 1  offset = 43008-40960 
I (1399) pmm_bitmap: TPmmBitMapLoad - 2 0x00002800  8
I (1399) nvm_like_eeprom: TNvmSdEepromRead 1  offset = 43016-40960 
I (1400) pmm_bitmap: TPmmBitMapCreate - i=136192  4  19456  0
I (1400) pmm_bitmap: TPmmBitMapLoad - 1 = 0x00004c00
I (1400) nvm_like_eeprom: TNvmSdEepromRead 1  offset = 52224-49152 
I (1401) pmm_bitmap: TPmmBitMapLoad - 2 0x00004c00  8
I (1401) nvm_like_eeprom: TNvmSdEepromRead 1  offset = 52232-49152 
I (1403) nvm_like_eeprom: TNvmSdEepromRead 1  offset = 53248-53248 
I (1404) nvm_like_eeprom: TNvmSdEepromRead 1  offset = 53256-53248 
I (1405) pmm_map: AstPmmGetPidActivePart - 1  133120-1-1
I (1405) pmm_bitmap: get_bit_value - 1 pid = 133120  133120  4  1
I (1406) pmm_bitmap: get_bit_value - 2 index = 0  0  0
I (1406) main: Frequency is 915MHz
I (1407) pmm_map: AstPmmGetPidActivePart - 1  132096-0-0
I (1407) pmm_bitmap: get_bit_value - 1 pid = 132096  132096  8  1
I (1407) pmm_bitmap: get_bit_value - 2 index = 0  0  0
I (1408) pmm_map: AstPmmGetPidActivePart - 1  133120-1-1
I (1408) pmm_bitmap: get_bit_value - 1 pid = 133120  133120  4  1
I (1408) pmm_bitmap: get_bit_value - 2 index = 0  0  0
I (1408) main: Frequency is 416062500MHz, LoraFre = 4, ChunFengVIN = 0x1234
I (1409) RA01S: CONFIG_MISO_GPIO=13
I (1409) RA01S: CONFIG_MOSI_GPIO=11
I (1409) RA01S: CONFIG_SCLK_GPIO=12
I (1409) RA01S: CONFIG_NSS_GPIO=10
I (1409) RA01S: CONFIG_RST_GPIO=8
I (1409) RA01S: CONFIG_BUSY_GPIO=9
I (1410) RA01S: CONFIG_TXEN_GPIO=15
I (1410) RA01S: CONFIG_RXEN_GPIO=7
I (1410) gpio: GPIO[10]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1411) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1411) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1411) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1412) gpio: GPIO[7]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1413) RA01S: spi_bus_initialize=0
I (1413) RA01S: spi_bus_add_device=0
I (1414) RA01S: LoRaBegin in1
I (1454) RA01S: Reset
I (1454) RA01S: syncWord=0x1424
I (1455) RA01S: SX126x installed1
W (1455) LORATEST: SetStandby1 MODE = 1
I (1455) RA01S: WriteCommand: CMD=0x80
E (1455) RA01S: in3 & 0b00001110a
E (1456) RA01S: SPI Transaction error:a

This is a retyped log, which contains additional component logs

niubbbbbbbb, 2023-02-10:
[image]

nopnop2002, 2023-02-10:
Are you using this?

https://github.com/nopnop2002/esp-idf-sx126x/tree/main/basic

niubbbbbbbb, 2023-02-10:
yes

niubbbbbbbb, 2023-02-10:
I used lierda's library is success

nopnop2002, 2023-02-10:
I don't know lierda's library.

# Option with SX1262/1268
LoRa modules with SX1262/1268 have several options.   

- Using TCXO(Temperature-Compensated Crystal Oscillator)   
SX1262/1268 can use the TCXO.   
If the TCXO is used, the XTB pin is not connected.   
However, the 6th pin (DIO3) of the SX1262/1268 can be used to power the TCXO.   
Explanation for TXCO and antenna control is [here](https://github.com/beegee-tokyo/SX126x-Arduino).   
Ra-01S / Ra-01SH does not use TCXO.   

- Power supply modes   
SX1262/1268 has two power supply modes.   
One is that only LDO used in all modes.   
Another is that DC_DC+LDO used for STBY_XOSC,FS, RX and TX modes.   
Explanation for LDO and DCDC selection is [here](https://github.com/beegee-tokyo/SX126x-Arduino).   
Ra-01S / Ra-01SH use only LDO in all modes.

- RF-Switching   
In general, use DIO2 to switch the RF-Switch.   
However, some tranceiver use an external gpio to switch the RF-Switch.   
Ra-01S / Ra-01SH use the SC70-6 integrated load switch to switch between RFO and RFI.   
Ra-01S / Ra-01SH use DIO2 to control this.   
DIO2 = 1, CTRL = 0, RFC to RF1  Tx Mode.   
DIO2 = 0, CTRL = 1, RFC to R21  Rx Mode.  

You need to look at the schematic to set these options properly, but it's very esoteric.   
The default settings for this library are for Ra-01S / Ra-01SH.    
__When using other than Ra-01S / Ra-01SH, you need to set them appropriately.__   
I don't have a SX126X other than Ra-01S / Ra-01SH.   
Therefore, I don't know the settings of SX126X other than Ra-01S / Ra-01SH.

niubbbbbbbb, 2023-02-11:
It's not about setting up TCXO. This error occurs in setstandby before setting TCXO

nopnop2002, 2023-02-11:
There are no errors in my log.

nopnop2002, 2023-02-11:
Your logging is completely different from mine.

I will ask again.

are you really doing this using esp-idf?

https://github.com/nopnop2002/esp-idf-sx126x/tree/main/basic

niubbbbbbbb, 2023-02-11:
Yes, I did use components under that path. I added the red log manually. I developed it using vscode and ESP-IDF

nopnop2002, 2023-02-11:
Can you run idf.py from the command line?

```
idf.py --version
```

niubbbbbbbb, 2023-02-11:
4.4.3

nopnop2002, 2023-02-11:
OK

try this command
```
git clone https://github.com/nopnop2002/esp-idf-sx126x
cd esp-idf-sx126x/basic/
idf.py set-target {esp32/esp32s2/esp32s3/esp32c2/esp32c3}
idf.py menuconfig
idf.py flash monitor
```

niubbbbbbbb, 2023-02-11:
Sorry, if menuconfig is modified, my project will not run normally

nopnop2002, 2023-02-11:
>if menuconfig is modified, my project will not run normally

Why?

Here are the official build instructions.

niubbbbbbbb, 2023-02-11:
Lora is only a part of my project, and there are other components that need to change menuconfig to run. I did not upload this project to gitlab, so I am worried that the project cannot run normally after modification.

nopnop2002, 2023-02-11:
>Lora is only a part of my project, and there are other components that need to change menuconfig to run.

Even if your project doesn't work
That's your problem, not my library's.

Please do not report problems here.

closed 2023-02-11 by nopnop2002

niubbbbbbbb, 2023-02-11:
No, your library should also have problems, because your library can run SX1262 but not LLCC68. Therefore, it is unlikely that any of my other components caused it. After the project is completed, I will find time to investigate this problem

nopnop2002, 2023-02-11:
This library works fine on my LLCC68.