nopnop2002/esp-idf-sx126x #21: SetRx Illegal Status

654955321, 2023-09-21:
Hello, I am using the E22-900M30S module. How to solve the issue of error reporting in logs：

I (0) cpu_start: App cpu up.
I (197) cpu_start: Pro cpu start user code
I (197) cpu_start: cpu freq: 240000000 Hz
I (197) cpu_start: Application information:
I (200) cpu_start: Project name:     main
I (205) cpu_start: App version:      1
I (209) cpu_start: Compile time:     Sep 21 2023 15:41:03
I (215) cpu_start: ELF file SHA256:  b38a34ba74bfb5d3...
I (221) cpu_start: ESP-IDF:          v5.1.1
I (226) cpu_start: Min chip rev:     v0.0
I (231) cpu_start: Max chip rev:     v0.99
I (236) cpu_start: Chip rev:         v0.1
I (240) heap_init: Initializing. RAM available for dynamic allocation:
I (248) heap_init: At 3FC95E70 len 000538A0 (334 KiB): DRAM
I (254) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (261) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (267) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (274) spi_flash: detected chip: gd
I (277) spi_flash: flash io: dio
W (281) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (294) sleep: Configure to isolate all GPIO pins in sleep state
I (301) sleep: Enable automatic switching of GPIO sleep configuration
I (308) app_start: Starting scheduler on CPU0
I (313) app_start: Starting scheduler on CPU1
I (313) main_task: Started on CPU0
I (323) main_task: Calling app_main()
I (323) MAIN: Frequency is 915MHz
I (333) RA01S: CONFIG_MISO_GPIO=12
I (333) RA01S: CONFIG_MOSI_GPIO=11
I (343) RA01S: CONFIG_SCLK_GPIO=10
I (343) RA01S: CONFIG_NSS_GPIO=9
I (343) RA01S: CONFIG_RST_GPIO=13
I (353) RA01S: CONFIG_BUSY_GPIO=14
I (353) RA01S: CONFIG_TXEN_GPIO=-1
I (363) RA01S: CONFIG_RXEN_GPIO=-1
I (363) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (373) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (383) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (393) RA01S: spi_bus_initialize=0
I (393) RA01S: spi_bus_add_device=0
I (443) RA01S: Reset
I (443) RA01S: syncWord=0x1424
I (443) RA01S: SX126x installed
W (443) RA01S: WriteCommand2 status=0a retry=1
I (443) RA01S: tcxoVoltage=0.000000
I (453) RA01S: useRegulatorLDO=0
W (453) RA01S: WriteCommand2 status=0a retry=1
W (463) RA01S: WriteCommand2 status=0a retry=1
I (463) RA01S: SetStopRxTimerOnPreambleDetect enable=0
E (483) RA01S: SetRx Illegal Status

nopnop2002, 2023-09-21:
//float tcxoVoltage = 0.0; // don't use TCXO
	//bool useRegulatorLDO = false; // use only LDO in all modes

	float tcxoVoltage = 3.3; // use TCXO
	bool useRegulatorLDO = true; // use TCXO

654955321, 2023-09-21:
I tried to modify it like this before, but the displayed log is different from your README, and my receiving end cannot receive it. The logs displayed after modification are as follows:
SENDER:
[image]
RECEIVER:
[image]