nopnop2002/esp-idf-sx126x #23: RA01S: WaitForIdle Timeout timeout=5000

pluzmedia, 2024-05-09:
Hi,

Encountered the "RA01S: WaitForIdle Timeout timeout=5000" error randomly during operation. How to restart it?

Thanks

nopnop2002, 2024-05-09:
I don't know because that message is not coming from this library.

```
$ cd esp-idf-sx126x
$ grep -rn "WaitForIdle" *　---> nowhere
```

pluzmedia, 2024-05-09:
The problem appears randomly and i have checked the wring and there was no conflict.

pluzmedia, 2024-05-09:
Will it cause issues when sending and receiving data every 50 milliseconds?

nopnop2002, 2024-05-09:
A problem may occur with the single-core ESP32-C series.
This is a multitasking constraint.

pluzmedia, 2024-05-09:
I am using a dual-core series, and apart from the error message E (68985) RA01S: WaitForIdle Timeout timeout=10000, no logs were printed.

nopnop2002, 2024-05-09:
The main task may not have core execution privileges assigned for a long time.

pluzmedia, 2024-05-09:
Are there any workarounds for cases like restarting or resetting the LoRa module or something similar?

nopnop2002, 2024-05-09:
ESP-IDF has an API for esp_restart, but even if you restart or reset, the problem will not be resolved.

Tasks (using SX126X) may not have core execution privileges assigned for a long time, and this needs to be resolved.

pluzmedia, 2024-05-09:
`Tasks (using SX126X) may not have core execution privileges assigned for a long time, and this needs to be resolved.` throwing some light would be helpful.

nopnop2002, 2024-05-09:
>throwing some light would be helpful.

I can't understand your English.

pluzmedia, 2024-05-09:
Any ideas on how to resolve the issue you mentioned earlier would be helpful.

nopnop2002, 2024-05-09:
for example. ．． ．． ．． ．．
Check the priority of each task.
Find out how much time each task is consuming.

pluzmedia, 2024-05-09:
Thanks, I will look into it.

nopnop2002, 2024-05-09:
Note that in ESP-IDF each task occupies at least 10ms of CPU.

pluzmedia, 2024-05-09:
I increased the delay of other tasks; let's see how it works. It's a complicated project. Thank you for your valuable insight.

nopnop2002, 2024-05-09:
Since ESP-IDF is a multitasking environment, it is necessary to pay attention to the CPU occupation time and priority of each task, which may affect operation.

pluzmedia, 2024-05-19:
I removed the WaitForIdleTimeout, and it works fine. Thanks for your timely support. I also have one more question: What does a bandwidth of 4 mean? Which kHz does it correspond to?

nopnop2002, 2024-05-19:
>What does a bandwidth of 4 mean?

Everything is in the datasheet.

[image]

pluzmedia, 2024-05-20:
Thank you, Will txPowerInDbm has any impact on range? I changed the value from 1 to 22 but it has zero impact on range.

nopnop2002, 2024-05-21:
txPowerInDbm() updates register 0x8E.
Please check the data sheet of the module you are using to determine the operation of this register.

pluzmedia, 2024-05-21:
Thanks to your support, I was able to achieve my goal. Initially, using SF7, I obtained a range of 200 meters in an urban area, which was disappointing. I decided to increase the TX power, but it had no impact. After increasing the SF to 11, I achieved a range of 700 meters in an urban area densely surrounded by buildings. However, the problem with SF11 is the restriction on sending data in a single packet. To overcome this, I chunked the data into smaller bytes and transmitted them to the receiver side. At the receiver side, the data is reassembled and decrypted to obtain the complete data, which I was previously sending using SF7. The only tradeoff between SF7 and SF11 is the airtime, which is about one second. This setup works well for communication with five nodes. Thank you for your kind and timely support in helping me achieve this. Without your assistance, it would have been very difficult to complete on time.

nopnop2002, 2024-05-22:
In LoRa modulation mode, the communication speed (bps) and maximum reception sensitivity (RSSI) are determined by a combination of spreading factor (SF), bandwidth (BW), and coding rate (CDR).
◆SF
SF (spreading factor) Increasing SF increases the spreading rate and improves noise resistance.
This improves reception sensitivity and extends communication distance, but communication speed decreases.
◆BW
Bandwidth sets the width of the communication band. Setting a larger bandwidth will improve communication speed.
However, the radio reception sensitivity (RSSI) will decrease.
◆CDR
CDR (CodingRate) sets the level of error correction rate. The larger the number, the better the correction rate, but the amount of information per packet increases.
Increase. (No effect on maximum reception sensitivity)
You can set whether to use Optimaise for each CDR, and enabling it will improve the correction rate, but will reduce communication speed.

closed 2024-05-22 by pluzmedia

nopnop2002, 2024-08-28:
I found out the cause of the Timeout error in WaitForIdle.