nopnop2002/esp-idf-sx126x #26: Watchdog timeout

h4alo, 2024-09-04:
W (402) MAIN: Disable TCXO
E (5322) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (5322) task_wdt:  - IDLE0 (CPU 0)
E (5322) task_wdt: Tasks currently running:
E (5322) task_wdt: CPU 0: main
E (5322) task_wdt: CPU 1: IDLE1
E (5322) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x4200E73A:0x3FC93960 0x4200EB5C:0x3FC93980 0x40376DB9:0x3FC939B0 0x4200A9DD:0x3FC995E0 0x42008E4A:0x3FC99600 0x42008F09:0x3FC99630 0x420095EB:0x3FC99650 0x42008AC7:0x3FC996A0 0x4201C293:0x3FC996D0 0x4037CCE5:0x3FC99700
0x4200e73a: task_wdt_timeout_handling at E:/esp/Espressif/frameworks/esp-idf-v5.1.4/components/esp_system/task_wdt/task_wdt.c:489 (discriminator 3)
0x4200eb5c: task_wdt_isr at E:/esp/Espressif/frameworks/esp-idf-v5.1.4/components/esp_system/task_wdt/task_wdt.c:565
0x40376db9: _xt_lowint1 at E:/esp/Espressif/frameworks/esp-idf-v5.1.4/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1240
0x4200a9dd: gpio_ll_get_level at E:/esp/Espressif/frameworks/esp-idf-v5.1.4/components/hal/esp32s3/include/hal/gpio_ll.h:375
 (inlined by) gpio_get_level at E:/esp/Espressif/frameworks/esp-idf-v5.1.4/components/driver/gpio/gpio.c:245
0x42008e4a: WaitForIdle at E:/esp/project/llcc68_audio/components/ra01s/ra01s.c:837
0x42008f09: Reset at E:/esp/project/llcc68_audio/components/ra01s/ra01s.c:438
0x420095eb: LoRaBegin at E:/esp/project/llcc68_audio/components/ra01s/ra01s.c:185
0x42008ac7: app_main at E:/esp/project/llcc68_audio/main/main.c:105
0x4201c293: main_task at E:/esp/Espressif/frameworks/esp-idf-v5.1.4/components/freertos/app_startup.c:208 (discriminator 13)
0x4037cce5: vPortTaskWrapper at E:/esp/Espressif/frameworks/esp-idf-v5.1.4/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

nopnop2002, 2024-09-04:
Which sample code did you launch?

h4alo, 2024-09-04:
> Which sample code did you launch?

I quoted the receive example, the send example doesn't give an error

nopnop2002, 2024-09-04:
Which project did you use?

closed 2024-11-05 by nopnop2002