nopnop2002/esp-idf-sx126x #27: SetRX Illegal Status (LoraErrorDefault = 18) using LLCC68 Tranceiver with ESP32-S3

ThiSo, 2024-11-01:
Hello!
I'm trying to run a basic example of LoRa Send using a ESP32-S3 in a board integrated with a LLCC68 tranceiver.
I'm getting an SetRX Illegal Status error, as can be seen in the log as follows:

```
I (26) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (27) boot: compile time Nov  1 2024 15:18:48
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0cd50h ( 52560) map
I (109) esp_image: segment 1: paddr=0001cd78 vaddr=3fc94200 size=02b2ch ( 11052) load
I (112) esp_image: segment 2: paddr=0001f8ac vaddr=40374000 size=0076ch (  1900) load
I (117) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1cf98h (118680) map
I (146) esp_image: segment 4: paddr=0003cfc0 vaddr=4037476c size=0fa18h ( 64024) load
I (167) boot: Loaded app from partition at offset 0x10000
I (167) boot: Disabling RNG early entropy source...
I (179) cpu_start: Multicore app
I (189) cpu_start: Pro cpu start user code
I (189) cpu_start: cpu freq: 160000000 Hz
I (189) app_init: Application information:
I (192) app_init: Project name:     lora
I (197) app_init: App version:      7d2d3da-dirty
I (202) app_init: Compile time:     Nov  1 2024 15:18:40
I (208) app_init: ELF file SHA256:  1185bbbc7...
I (213) app_init: ESP-IDF:          v5.3.1-dirty
I (219) efuse_init: Min chip rev:     v0.0
I (223) efuse_init: Max chip rev:     v0.99 
I (228) efuse_init: Chip rev:         v0.1
I (233) heap_init: Initializing. RAM available for dynamic allocation:
I (240) heap_init: At 3FC97618 len 000520F8 (328 KiB): RAM
I (246) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (253) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (259) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (266) spi_flash: detected chip: gd
I (269) spi_flash: flash io: dio
W (273) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (286) sleep: Configure to isolate all GPIO pins in sleep state
I (293) sleep: Enable automatic switching of GPIO sleep configuration
I (301) main_task: Started on CPU0
I (311) main_task: Calling app_main()
I (311) MAIN: Frequency is 915MHz
I (311) gpio: GPIO[35]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (321) LLCC68: spi_bus_initialize=0
I (331) LLCC68: spi_bus_add_device=0
I (331) LLCC68: Reset
I (331) LLCC68: ReadRegister: REG=0x740
I (341) LLCC68: DataIn:14 
I (341) LLCC68: DataIn:24 
I (341) LLCC68: syncWord=0x1424
I (351) LLCC68: SX126x installed
I (351) LLCC68: WriteCommand: CMD=0x80
I (361) LLCC68: 00 --> aa
W (361) LLCC68: WriteCommand2 status=0a retry=1
I (361) LLCC68: WriteCommand: CMD=0x80
I (371) LLCC68: 00 --> a2
I (371) LLCC68: WriteCommand: CMD=0x9d
I (381) LLCC68: 01 --> a2
I (381) LLCC68: tcxoVoltage=0.000000
I (381) LLCC68: WriteCommand: CMD=0x89
I (391) LLCC68: 7f --> a2
I (391) LLCC68: useRegulatorLDO=0
I (401) LLCC68: WriteCommand: CMD=0x96
I (401) LLCC68: 01 --> aa
W (401) LLCC68: WriteCommand2 status=0a retry=1
I (411) LLCC68: WriteCommand: CMD=0x96
I (411) LLCC68: 01 --> a2
I (421) LLCC68: WriteCommand: CMD=0x8f
I (421) LLCC68: 00 --> a2
I (421) LLCC68: 00 --> a2
I (431) LLCC68: WriteCommand: CMD=0x95
I (431) LLCC68: 04 --> a2
I (441) LLCC68: 07 --> a2
I (441) LLCC68: 00 --> a2
I (441) LLCC68: 01 --> a2
I (451) LLCC68: WriteRegister: REG=0x8e7
I (451) LLCC68: 18 --> a2
I (451) LLCC68: WriteCommand: CMD=0x8e
I (461) LLCC68: 00 --> a2
I (461) LLCC68: 04 --> a2
I (461) LLCC68: WriteCommand: CMD=0x98
I (471) LLCC68: e1 --> a2
I (471) LLCC68: e9 --> a2
I (481) LLCC68: WriteCommand: CMD=0x86
I (481) LLCC68: 39 --> aa
W (481) LLCC68: WriteCommand2 status=0a retry=1
I (491) LLCC68: WriteCommand: CMD=0x86
I (491) LLCC68: 39 --> a2
I (501) LLCC68: 38 --> a2
I (501) LLCC68: 00 --> a2
I (501) LLCC68: 00 --> a2
I (511) LLCC68: SetStopRxTimerOnPreambleDetect enable=0
I (511) LLCC68: WriteCommand: CMD=0x9f
I (521) LLCC68: 00 --> a2
I (521) LLCC68: WriteCommand: CMD=0xa0
I (521) LLCC68: 00 --> a2
I (531) LLCC68: SetLoRaSymbNumTimeout Done
I (531) LLCC68: WriteCommand: CMD=0x8a
I (541) LLCC68: 01 --> a2
I (541) LLCC68: SetPacketType Done
I (541) LLCC68: WriteCommand: CMD=0x8b
I (551) LLCC68: 0b --> a2
I (551) LLCC68: 06 --> a2
I (551) LLCC68: 04 --> a2
I (561) LLCC68: 00 --> a2
I (561) LLCC68: SetModulationParams Done
I (571) LLCC68: ReadRegister: REG=0x736
I (571) LLCC68: DataIn:0d 
I (571) LLCC68: WriteRegister: REG=0x736
I (581) LLCC68: 0d --> a2
I (581) LLCC68: FixInvertedIQ Done
I (591) LLCC68: WriteCommand: CMD=0x8c
I (591) LLCC68: 00 --> a2
I (591) LLCC68: 08 --> a2
I (601) LLCC68: 00 --> a2
I (601) LLCC68: ff --> a2
I (601) LLCC68: 01 --> a2
I (611) LLCC68: 00 --> a2
I (611) LLCC68: SetPacketParams Done
I (611) LLCC68: WriteCommand: CMD=0x08
I (621) LLCC68: 03 --> a2
I (621) LLCC68: ff --> a2
I (631) LLCC68: 00 --> a2
I (631) LLCC68: 00 --> a2
I (631) LLCC68: 00 --> a2
I (641) LLCC68: 00 --> a2
I (641) LLCC68: 00 --> a2
I (641) LLCC68: 00 --> a2
I (651) LLCC68: SetDioIrqParams Done
I (651) LLCC68: ----- SetRx timeout=16777215
I (651) LLCC68: WriteCommand: CMD=0x80
I (661) LLCC68: 00 --> a2
I (661) LLCC68: SetRxEnable:LLCC68_TXEN=-1 LLCC68_RXEN=-1
I (671) LLCC68: SetRxEnable Done
I (671) LLCC68: WriteCommand: CMD=0x82
I (681) LLCC68: ff --> a2
I (681) LLCC68: ff --> a2
I (681) LLCC68: ff --> a2
I (691) LLCC68: ReadCommand: CMD=0xc0
I (691) LLCC68: DataIn:2a
I (701) LLCC68: ReadCommand: CMD=0xc0
I (701) LLCC68: DataIn:2a
I (701) LLCC68: ReadCommand: CMD=0xc0
I (711) LLCC68: DataIn:2a
I (711) LLCC68: ReadCommand: CMD=0xc0
I (711) LLCC68: DataIn:2a
I (721) LLCC68: ReadCommand: CMD=0xc0
I (721) LLCC68: DataIn:2a
I (731) LLCC68: ReadCommand: CMD=0xc0
I (731) LLCC68: DataIn:2a
I (731) LLCC68: ReadCommand: CMD=0xc0
I (741) LLCC68: DataIn:2a
I (741) LLCC68: ReadCommand: CMD=0xc0
I (741) LLCC68: DataIn:2a
I (751) LLCC68: ReadCommand: CMD=0xc0
I (751) LLCC68: DataIn:2a
I (761) LLCC68: ReadCommand: CMD=0xc0
I (761) LLCC68: DataIn:2a
I (761) LLCC68: ReadCommand: CMD=0xc0
I (771) LLCC68: DataIn:2a
E (771) LLCC68: SetRx Illegal Status
E (771) LLCC68: LoRaErrorDefault=18
```

I already tested enabling TCXO and changing the power supply mode, but the error persists.
And I also think my pins are correctly connected (except for BUSY, which my ESP32-S3 doesn't have, but I tested removing it from the code in an ESP32 where it was working and it had no real effect).

Any ideas what else can I try/change to make it work?
Thanks in advance!

nopnop2002, 2024-11-01:
>except for BUSY, which my ESP32-S3 doesn't have

Using menuconfig, BUSY can be changed to any GPIO of your choice.

s26389057, 2024-11-05:
Hello, I am a beginner and I would like to know the reason? I am using a LORA module that combines E220-900M22S with ESP32S3 according to the schematic MCU IO NSS-IO7 SCK-I05 MOSI-I06 MISO-IOS DI01-IO33 BUSY-I034 NRST-IO8 only has these interfaces. I want to know where the mistake lies?

`I (26) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (27) boot: compile time Nov  5 2024 14:09:48
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0d050h ( 53328) map
I (109) esp_image: segment 1: paddr=0001d078 vaddr=3fc94200 size=02b2ch ( 11052) load
I (112) esp_image: segment 2: paddr=0001fbac vaddr=40374000 size=0046ch (  1132) load
I (117) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=204e4h (132324) map
I (149) esp_image: segment 4: paddr=0004050c vaddr=4037446c size=0fd18h ( 64792) load
I (170) boot: Loaded app from partition at offset 0x10000
I (170) boot: Disabling RNG early entropy source...
I (182) cpu_start: Multicore app
I (191) cpu_start: Pro cpu start user code
I (191) cpu_start: cpu freq: 160000000 Hz
I (191) app_init: Application information:
I (194) app_init: Project name:     lora
I (199) app_init: App version:      v5.3.1-dirty
I (204) app_init: Compile time:     Nov  5 2024 14:09:28
I (210) app_init: ELF file SHA256:  81fb27777...
I (215) app_init: ESP-IDF:          v5.3.1-dirty
I (221) efuse_init: Min chip rev:     v0.0
I (225) efuse_init: Max chip rev:     v0.99
I (230) efuse_init: Chip rev:         v0.2
I (235) heap_init: Initializing. RAM available for dynamic allocation:
I (242) heap_init: At 3FC97628 len 000520E8 (328 KiB): RAM
I (248) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (255) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (261) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (268) spi_flash: detected chip: generic
I (272) spi_flash: flash io: dio
W (276) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (289) sleep: Configure to isolate all GPIO pins in sleep state
I (296) sleep: Enable automatic switching of GPIO sleep configuration
I (303) main_task: Started on CPU0
I (313) main_task: Calling app_main()
I (313) MAIN: Frequency is 433MHz
I (313) RA01S: CONFIG_MISO_GPIO=3
I (323) RA01S: CONFIG_MOSI_GPIO=6
I (323) RA01S: CONFIG_SCLK_GPIO=5
I (323) RA01S: CONFIG_NSS_GPIO=7
I (333) RA01S: CONFIG_RST_GPIO=8
I (333) RA01S: CONFIG_BUSY_GPIO=34
I (343) RA01S: CONFIG_TXEN_GPIO=5
I (343) RA01S: CONFIG_RXEN_GPIO=4
I (343) gpio: GPIO[7]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (353) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (363) gpio: GPIO[34]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (373) gpio: GPIO[5]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (383) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (393) RA01S: spi_bus_initialize=0
I (393) RA01S: spi_bus_add_device=0
W (403) MAIN: Enable TCXO
I (443) RA01S: Reset
I (443) RA01S: syncWord=0x1424
I (443) RA01S: SX126x installed
I (443) RA01S: tcxoVoltage=3.300000
I (453) RA01S: useRegulatorLDO=1
W (453) RA01S: WriteCommand2 status=0a retry=1
W (463) RA01S: WriteCommand2 status=0a retry=1
I (463) RA01S: SetStopRxTimerOnPreambleDetect enable=0
E (583) RA01S: SetRx Illegal Status`

ThiSo, 2024-11-05:
> Using menuconfig, BUSY can be changed to any GPIO of your choice. 

> 

Sorry, I meant to say that the LLCC68 BUSY pin isn't really soldered to any GPIO in my ESP32-S3, so it is set to -1 in menuconfig. This worked when using my other ESP32 (image below, where the last brown connector is for BUSY), so I don't think that's the issue.

[image]

My pin connection in both microcontrolers were as follows:

| PIN | ESP32  | ESP32-S3|
|-------| ------------- | ------------- |
| MISO |12 | 16 |
| MOSI  | 13  | 15 |
| SCLK | 14  | 17 |
| NSS  | 15  | 35 |
| BUSY | -1  | -1 |
| RST  | -1 | -1 |
| TXEN | -1  | -1 |
| RXEN | -1  | -1 |

> SX1262 and LLCC68 are compatible, but for some reason they don't work.
>

Also, I'm not using any EBYTE module, so I don't really understand if this table has to do with the problem. My chip is this one:

[image]

nopnop2002, 2024-11-05:
Sort out the problem.

ESP32+LLCC68 works normally.

ESP32S3+LLCC68 does not work properly.

Is this correct?

ThiSo, 2024-11-05:
Yes, that's right.

ThiSo, 2024-11-05:
> What happens if you enable TXEN and RXEN and connect them?
>

They are also not soldered. The only ones really connected to the ESP32S3 are the four enabled in the table above.

[image]

Sorry for the picture quality, but the LLCC68 is the one directly above the ESP32S3.

nopnop2002, 2024-11-05:
| PIN | ESP32  | ESP32-S3|
|-------| ------------- | ------------- |
| MISO |12 | 16 |
| MOSI  | 13  | 15 |
| SCLK | 14  | 17 |
| NSS  | 15  | 35 |
| BUSY | -1  | -1 |
| RST  | -1 | -1 |
| TXEN | -1  | -1 |
| RXEN | -1  | -1 |

In my sample code, you cannot change BUSY and RST GPIO to -1.
[image]

ThiSo, 2024-11-05:
I altered it in Kconfig.projbuild:

`       config RST_GPIO
		int "SX126X RST GPIO"
		range -1 GPIO_RANGE_MAX
		default 16 if IDF_TARGET_ESP32
		default 38 if IDF_TARGET_ESP32S2 || IDF_TARGET_ESP32S3
		default  2 # C3 and others
		help
			Pin Number to be used as the RST signal.

	config BUSY_GPIO
		int "SX126X BUSY GPIO"
		range -1 GPIO_RANGE_MAX
		default 17 if IDF_TARGET_ESP32
		default 39 if IDF_TARGET_ESP32S2 || IDF_TARGET_ESP32S3
		default  3 # C3 and others
		help
			Pin Number to be used as the BUSY signal.`

nopnop2002, 2024-11-05:
>I altered it in Kconfig.projbuild:
Output to GPIO=-1 occurs.
At the same time, code changes will also be required.

In the schematic you showed, LLCC_RESET and LLCC_BUSY are named, so they are connected somewhere.

ThiSo, 2024-11-05:
>At the same time, code changes will also be required.
>
Other changes I made for it to work on ESP32 are as follows:

In LoRaInit()
```
if (LLCC68_RST != -1) {
		gpio_reset_pin(LLCC68_RST);
		gpio_set_direction(LLCC68_RST, GPIO_MODE_OUTPUT);
	}
	
if (LLCC68_BUSY != -1) {
		gpio_reset_pin(LLCC68_BUSY);
		gpio_set_direction(LLCC68_BUSY, GPIO_MODE_INPUT);
}
```
	
In Reset():

```
if (LLCC68_RST != -1) {
		delay(10);
		gpio_set_level(LLCC68_RST,0);
		delay(20);
		gpio_set_level(LLCC68_RST,1);
		delay(10);
}
```

nopnop2002, 2024-11-05:
I can't tell if your changes are correct.

closed 2024-11-05 by nopnop2002