nopnop2002/esp-idf-sx126x #29: Receiving Meshtastic Packets

AkaBkn, 2025-02-10:
I'm using SX1262 in LilyGO T-Deck Plus.

For a test of reception, I try to receive packets from meshtastic.

so i set EU_868 LongFast params like that 
```
#define SX126X_LORA_SF11 0x0b
uint8_t spreadingFactor = SX126X_LORA_SF11;
uint8_t bandwidth = SX126X_LORA_BW_250_0;
uint8_t codingRate = SX126X_LORA_CR_4_5;
uint16_t preambleLength = 16;
uint8_t payloadLen = 0;
int16_t syncword = 0x2b;
frequencyInHz = 869525000;
```

after LoRaConfig()
```
SetSyncWord(syncword);
```

there is no error in the serial log, however, i never receive any packets. if I trace the `irqRegs` in `LoRaReceive` i see that i get `SX126X_IRQ_PREAMBLE_DETECTED` but never `SX126X_IRQ_RX_DONE`.

have you an idea what i do wrong? do i need to configure DIO1 and handle interrupts like that?

nopnop2002, 2025-02-10:
There is no information about LilyGO T-Deck Plus.

You need to look at the schematic to set these options properly.

- Using TCXO(Temperature-Compensated Crystal Oscillator)   
SX1262/1268 can use the TCXO.   
If the TCXO is used, the XTB pin is not connected.   
However, the 6th pin (DIO3) of the SX1262/1268 can be used to power the TCXO.   
Explanation for TXCO and antenna control is [here](https://github.com/beegee-tokyo/SX126x-Arduino).   
Ra-01S / Ra-01SH does not use TCXO.   

- Power supply modes   
SX1262/1268 has two power supply modes.   
One is that only LDO used in all modes.   
Another is that DC_DC+LDO used for STBY_XOSC,FS, RX and TX modes.   
Explanation for LDO and DCDC selection is [here](https://github.com/beegee-tokyo/SX126x-Arduino).   
Ra-01S / Ra-01SH use only LDO in all modes.

- RF-Switching   
In general, use DIO2 to switch the RF-Switch.   
However, some tranceiver use an external gpio to switch the RF-Switch.   
Ra-01S / Ra-01SH use the SC70-6 integrated load switch to switch between RFO and RFI.   
Ra-01S / Ra-01SH use DIO2 to control this.   
DIO2 = 1, CTRL = 0, RFC for SC70-6 goes to RF1 -->  Switch to Tx Mode.   
DIO2 = 0, CTRL = 1, RFC for SC70-6 gose to RF2 -->  Switch to Rx Mode.

AkaBkn, 2025-02-10:
Meshtastic (https://meshtastic.org/) is a LoRa mesh networking system.

TCXO / LDO: pretty sure i have those correct, also played with all the combinations.

RF switching: on the schematics, (https://github.com/Xinyuan-LilyGO/T-Deck/blob/master/schematic/schematic.pdf) i only see DIO1 connected.

`payloadLen = 0;` meens no fixed lenght if I'm not mistaken. Should be cirrect.

Why would it receive the preamble but not the packet?

nopnop2002, 2025-02-10:
Do the Header Modes of both match?

AkaBkn, 2025-02-10:
Yes, `payloadLen = 0` should mean explicit, which should match.

nopnop2002, 2025-02-10:
Will it work properly using two LilyGO T-Deck Plus?

AkaBkn, 2025-02-10:
Unfortunatly I have only one. I send meshtastic messages from a LilyGO T-Echo, it has the same SX1262 LoRa chip in there.

nopnop2002, 2025-02-10:
It seems that the LoRa packet formats do not match.

AkaBkn, 2025-02-10:
some other guy also has the same parameters (https://www.reddit.com/r/meshtastic/comments/1ec1us8/meshtastic_packets_not_received_by_a_standard/)

how would you change the packet header length?
is `SX126X_REG_PACKECT_LENGTH` also important?

AkaBkn, 2025-02-10:
got it! it was the wrong sync word.
meshtastic is using radiolib, which sets the syncword like that:

in src/modules/SX126x/SX126x.h
```
   786      \param controlBits Undocumented control bits, required for compatibility purposes.
...
   789    int16_t setSyncWord(uint8_t syncWord, uint8_t controlBits = 0x44);
```

in src/modules/SX126x/SX126x.cpp
```
   815    const uint8_t data[2] = {(uint8_t)((syncWord & 0xF0) | ((controlBits & 0xF0) >> 4)), (uint8_t)(((syncWord & 0x0F) << 4) | (controlBits & 0x0F))};
   816    return(writeRegister(RADIOLIB_SX126X_REG_LORA_SYNC_WORD_MSB, data, 2));
```

so when using `0x2b` as sync word, actually `0x24b4` gets written to the register.

maybe you want to also change `SetSyncWord` or add a note ?

nopnop2002, 2025-02-10:
https://github.com/nopnop2002/esp-idf-sx126x/blob/main/components/ra01s/ra01s.c#L614

closed 2025-02-25 by AkaBkn