nopnop2002/esp-idf-sx126x #31: SPI Transactions with hardware CS control [enhancement]

AkaBkn, 2025-02-11:
While using your library, I noticed that you do software CS control. 
As this creates problems when using multiple SPI devices, i rewrote it to use hardware CS control.
Tell me if i should make a pull request.

nopnop2002, 2025-02-11:
Could you conduct a long-term continuous operation test?

If possible, continuous operation for about 24 hours is desirable.

nopnop2002, 2025-02-14:
@AkaBkn 

I remembered this comment.

https://github.com/nopnop2002/esp-idf-sx126x/blob/main/components/ra01s/ra01s.c#L113

I tried changing it but it doesn't work properly.

Can you share your code?

AkaBkn, 2025-02-17:
for now my code is not really polished, but it does the job. 
its essential that you make only one SPI transfer for a command / read / write register.
so you need to rewrite all the essential functions and discard your `uint8_t spi_transfer(uint8_t address)`

see this for example:
```
void ReadRegister(uint16_t reg, uint8_t* data, uint8_t numBytes) {
        // ensure BUSY is low (state meachine ready)
        WaitForIdle(BUSY_WAIT, "start ReadRegister", true);

        if(debugPrint) {
                ESP_LOGI(TAG, "ReadRegister: REG=0x%02x", reg);
        }

        uint8_t buf[16] = {
                SX126X_CMD_READ_REGISTER,
                (reg & 0xFF00) >> 8,
                reg & 0xff,
        };

        memset(&buf[3], SX126X_CMD_NOP, sizeof(buf) - 3);

        // send command byte
        spi_read_byte(buf, buf, 4 + numBytes);
        memcpy(data, &buf[4], numBytes);

        WaitForIdle(BUSY_WAIT, "end ReadRegister", false);
}

```

i have not checked about the max. for readregsister. so i just took 16 for now :)

similairly
```
uint8_t ReadCommand(uint8_t cmd, uint8_t* data, uint8_t numBytes) {
        // ensure BUSY is low (state meachine ready)
        WaitForIdleBegin(BUSY_WAIT, "start ReadCommand");

        // start transfer

        // send command byte
        if(debugPrint) {
                ESP_LOGI(TAG, "ReadCommand: CMD=0x%02x", cmd);
        }

        uint8_t buf[16];
        buf[0] = cmd;

        memset(&buf[1], SX126X_CMD_NOP, sizeof(buf) - 1);

        spi_read_byte(buf, buf, 2 + numBytes);

        if (data != NULL && numBytes)
                memcpy(data, &buf[2], numBytes);

        // wait for BUSY to go low
        vTaskDelay(1);
        WaitForIdle(BUSY_WAIT, "end ReadCommand", false);

        return buf[1];
}
```

tell me if you need more info..

nopnop2002, 2025-02-18:
thank you.
WriteCommand2 becomes difficult.

AkaBkn, 2025-02-18:
there you go
```
uint8_t WriteCommand2(uint8_t cmd, uint8_t* data, uint8_t numBytes) {
        // ensure BUSY is low (state meachine ready)
        WaitForIdle(BUSY_WAIT, "start WriteCommand2", true);

        // send command byte
        if(debugPrint) {
                ESP_LOGI(TAG, "WriteCommand: CMD=0x%02x", cmd);
        }

        uint8_t buf[16];
        buf[0] = cmd;
        memcpy(&buf[1], data, numBytes);

        spi_read_byte(buf, buf, numBytes + 1);

        uint8_t status = 0;
        uint8_t cmd_status = buf[1] & 0xe;

        switch(cmd_status){
                case SX126X_STATUS_CMD_TIMEOUT:
                case SX126X_STATUS_CMD_INVALID:
                case SX126X_STATUS_CMD_FAILED:
                        status = cmd_status;
                                break;

                case 0:
                case 7:
                        status = SX126X_STATUS_SPI_FAILED;
                        break;
                // default: break; // success
          }

        // wait for BUSY to go low
        WaitForIdle(BUSY_WAIT, "end WriteCommand2", false);

        return status;
}
```

nopnop2002, 2025-02-24:
Changed to SPI transactions controlled by hardware CS.

Thank you for all the help.

closed 2025-02-25 by AkaBkn