nopnop2002/esp-idf-sx126x #38: Interrupt implementation within library

shibcreate, 2025-07-30:
I wanted to look into supporting interrupts for this library for some application needs. Was this looked into before / How difficult do you suspect the driver change to be to support hardware interrupts?

nopnop2002, 2025-07-30:
As you know, ESP-IDF supports multitasking.

Each task can wait for events without using the CPU at all.

For this reason, I have never considered hardware interrupts until now.

If you read the datasheet carefully, it is possible to modify the driver to support hardware interrupts.

I don't know how difficult the modification would be.

shibcreate, 2025-07-30:
Understood! Thank you for the information. I will take a look and make a PR when ready.

shibcreate, 2025-07-31:
For sure, understood. I wanted to look into support FSK because it seems it could have a faster data rate. I'll take a look to see the advantages, but I'm interested in expanding the driver further to support the protocol.

nopnop2002, 2025-07-31:
I haven't tried it, but I think the following will notify the DIO1 port of a receive completion interrupt.

```
	uint16_t irqMask = SX126X_IRQ_RX_DONE;
	uint16_t dio1Mask = 1;
	uint16_t dio2Mask = 0;
	uint16_t dio3Mask = 0;

	SetDioIrqParams(irqMask, dio1Mask, dio2Mask, dio3Mask);
```

shibcreate, 2025-07-31:
Thank you! I'll test it and see.

nopnop2002, 2025-07-31:
I added interrupt project. Please try.

https://github.com/nopnop2002/esp-idf-sx126x/tree/main/interrupt

shibcreate, 2025-07-31:
Will test soon!

closed 2025-08-26 by nopnop2002